adafruit/Adafruit_CircuitPython_MCP230xx #10: Needs a refactor.

caternuson, 2019-05-03:
Everything is currently in a single file, which is causing some memory issues:
https://github.com/adafruit/Adafruit_CircuitPython_CharLCD/issues/36

Suggest this gets broken out into several files in a subfolder:

* `adafruit_mcp230xx`
  * `mcp230xx.py` - base class
  * `mcp23008.py` - MCP23008 class
  * `mcp23017.py` - MCP23017 class
  * `digital_inout.py` - the DigitalInOut shim class   

***NOTE: This will break existing code***
. Usage will have to change from this:
```python
import adafruit_mcp230xx
mcp = adafruit_mcp230xx.MCP23008(i2c)
```
to something like this:
```python
from adafruit_mcp230xx.mcp23008 import MCP23008
mcp = MCP23008(i2c)
```

ladyada, 2019-05-03:
kk!

caternuson, 2019-05-16:
@dherrada oops, sry, i had already chipped away at this. just PR'd

dherrada, 2019-05-16:
@caternuson That’s fine. I’ll do some testing of it once it gets merged.

caternuson, 2019-05-16:
Do you have some bare MCP ICs? That'd be great. I'm only testing with an LCD backpack.

dherrada, 2019-05-16:
I don't have the bare ones, but I do have the GPIO expander and the LCD backpack.

caternuson, 2019-05-16:
OK. All green with Sir Travis. Test away #11 

Note that the CharLCD library will not work with this refactor until it also gets updated. Very minor tweaks to the imports and MCP setup.

closed 2019-05-16 by dherrada

dherrada, 2019-05-25:
@caternuson Everything got kind of crazy the last week, but I've just been able to test it, and from what I can tell, it works.