adafruit/Adafruit_CircuitPython_MCP230xx #12: No module named 'adafruit_mcp230xx'

ecspresso, 2019-05-19:
**Steps to reproduce**

```
$ pip3 install adafruit-circuitpython-mcp230xx
$ python3
>>> import adafruit_mcp230xx
```

Expected behavior

Import of module `adafruit_mcp230xx`

Actual behavior

```
>>> import adafruit_mcp230xx
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named 'adafruit_mcp230xx'
```

Environment
**RPI:** Raspberry Pi B (model 1)
**OS:** Raspbian Stretch Lite (Release date: 2019-04-08)

---

I am trying to use a Raspberry Pi B with a Adafruit 16x2 Character LCD + Keypad for Raspberry Pi and following the instructions, need this library. Installing it via pip3 does not seem to install it. 
Am I doing something wrong?

vonnieda, 2019-05-20:
Just +1 on this. Having the same problem trying to do the same thing.

vonnieda, 2019-05-20:
Okay, just figured this out. I don't know if this is the correct solution, but it works for me:

```
python3 -m pip install adafruit-circuitpython-charlcd
wget https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/download/20190516/adafruit-circuitpython-bundle-py-20190516.zip
unzip adafruit-circuitpython-bundle-py-20190516.zip 
PYTHONPATH=/home/flash/lcd/adafruit-circuitpython-bundle-py-20190516/lib python3 lcd.py 
```

Note that I am using `python3 -m pip` rather than `pip3` just to be extra sure I am using the right version.

What is happening here is that installing `adafruit-circuitpython-mcp230x` does not seem to install the required `adafruit_mcp230xx` library, and that library does not seem to be available via pip. Downloading the CircuitPython library bundle and then referencing it does the trick. 

Also note that when grabbing the CircuitPython library bundle you have to make sure you get the -py and not the -mpy. The -mpy ones are compiled for MicroPython and don't work on Raspberry Pi.

ladyada, 2019-05-20:
I think as soon as @caternuson is around he'll be able to fix it, we did some improvements on the library and probably just needs a small tweak to regress

caternuson, 2019-05-21:
Sry. Probably broke something with the PyPi distro with the refactor (#11). Will investigate and fix.

caternuson, 2019-05-21:
OK. Should be fixed now. Please try the latest 2.0.1 release which should fix this. To install:
```console
sudo pip3 install --upgrade --force-reinstall adafruit_circuitpython_mcp230xx
```
and as a quick test:
```console 
pi@raspberrypi:~ $ python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from adafruit_mcp230xx.mcp23008 import MCP23008
>>>
```

vonnieda, 2019-05-21:
Hi @caternuson, this seems to fix the missing import, but the charlcd library still isn't working:

```
python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from adafruit_mcp230xx.mcp23017 import MCP23017
>>> 
```

```
user@host:~/lcd $ python3 lcd2.py 
Traceback (most recent call last):
  File "lcd2.py", line 15, in <module>
    lcd = character_lcd.Character_LCD_RGB_I2C(i2c, lcd_columns, lcd_rows)
  File "/home/flash/.local/lib/python3.5/site-packages/adafruit_character_lcd/character_lcd_rgb_i2c.py", line 87, in __init__
    self._mcp = adafruit_mcp230xx.MCP23017(i2c)
AttributeError: module 'adafruit_mcp230xx' has no attribute 'MCP23017'
```
Maybe this means the character LCD library needs an update too? This seems to be the offending line: https://github.com/adafruit/Adafruit_CircuitPython_CharLCD/blob/master/adafruit_character_lcd/character_lcd_rgb_i2c.py#L87

Thanks,
Jason

caternuson, 2019-05-21:
Yep. That should be fixed soon also:
https://github.com/adafruit/Adafruit_CircuitPython_CharLCD/pull/39

closed 2019-05-21 by caternuson

dopyoman, 2019-05-23:
I'm still getting the same error even after reinstalling the packages.

caternuson, 2019-05-23:
Which error? The original one related to the package: 
```
ImportError: No module named 'adafruit_mcp230xx'
```
or the one related to the need to update CharLCD:
```
AttributeError: module 'adafruit_mcp230xx' has no attribute 'MCP23017'
```

gshorten, 2019-05-30:
I am still getting the same error message.  I did the forced reinstall, the files are there.
AttributeError: module 'adafruit_mcp230xx' has no attribute 'MCP23017'

caternuson, 2019-05-30:
That's the error related to the CharLCD library, which is a separate library/issue. It will be broken until it is updated to match the changes done to this library.

Keep an eye on this PR:
https://github.com/adafruit/Adafruit_CircuitPython_CharLCD/pull/39

gshorten, 2019-05-31:
> adafruit/Adafruit_CircuitPython_CharLCD#39

ok, thanks.  Maybe I need to switch to a different display?  the app is a
sonos controller, which takes input from a jukebox wallbox to play
selections on the sonos music system.  the display shows the track and
artist playing. Is broken :-(

caternuson, 2019-05-31:
It should work again once the other library is updated.

The general issue is with other libraries that used this library (like the CharLCD library). There should be no issues using this library as shown in the examples to interact with an MCP230xx directly.