adafruit/Adafruit_CircuitPython_MCP230xx #18: AttributeError: module 'adafruit_mcp230xx' has no attribute 'MCP23017'

tinystarship, 2019-11-01:
Based on some searching, it looks like this was fixed, but I'm still having trouble in version 2.2. This is all I'm trying to do:

`import board
import busio
import adafruit_mcp230xx
i2c = busio.I2C(board.SCL, board.SDA)
mcp = adafruit_mcp230xx.MCP23017(i2c)`

and every time I run it, it fails saying no attribute. I just did a force-reinstall based on a comment I saw in a previous issue, but it still fails.

`pi@NASA1:~/NASA $ sudo pip3 install --upgrade --force-reinstall adafruit_circuitpython_mcp230xx
Collecting adafruit_circuitpython_mcp230xx
  Using cached https://www.piwheels.org/simple/adafruit-circuitpython-mcp230xx/adafruit_circuitpython_mcp230xx-2.2.0-py3-none-any.whl
Collecting adafruit-circuitpython-busdevice (from adafruit_circuitpython_mcp230xx)
  Using cached https://www.piwheels.org/simple/adafruit-circuitpython-busdevice/adafruit_circuitpython_busdevice-4.0.0-py3-none-any.whl
Collecting Adafruit-Blinka (from adafruit_circuitpython_mcp230xx)
  Using cached https://www.piwheels.org/simple/adafruit-blinka/Adafruit_Blinka-3.0.0-py3-none-any.whl
Collecting sysv-ipc; platform_system != "Windows" (from Adafruit-Blinka->adafruit_circuitpython_mcp230xx)
  Using cached https://www.piwheels.org/simple/sysv-ipc/sysv_ipc-1.0.0-cp35-cp35m-linux_armv7l.whl
Collecting spidev; sys_platform == "linux" (from Adafruit-Blinka->adafruit_circuitpython_mcp230xx)
  Using cached https://www.piwheels.org/simple/spidev/spidev-3.4-cp35-cp35m-linux_armv7l.whl
Collecting Adafruit-PureIO (from Adafruit-Blinka->adafruit_circuitpython_mcp230xx)
  Using cached https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-0.2.3-py3-none-any.whl
Collecting RPi.GPIO (from Adafruit-Blinka->adafruit_circuitpython_mcp230xx)
  Using cached https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.0-cp35-cp35m-linux_armv7l.whl
Collecting Adafruit-PlatformDetect (from Adafruit-Blinka->adafruit_circuitpython_mcp230xx)
  Using cached https://www.piwheels.org/simple/adafruit-platformdetect/Adafruit_PlatformDetect-1.3.4-py3-none-any.whl
Collecting rpi-ws281x>=4.0.0 (from Adafruit-Blinka->adafruit_circuitpython_mcp230xx)
  Using cached https://files.pythonhosted.org/packages/32/99/1140d72d58e31d1251c2009238473e818f2d991e130c59f09dc7502d99bc/rpi_ws281x-4.2.2-cp35-cp35m-linux_armv7l.whl
Installing collected packages: sysv-ipc, spidev, Adafruit-PureIO, RPi.GPIO, Adafruit-PlatformDetect, rpi-ws281x, Adafruit-Blinka, adafruit-circuitpython-busdevice, adafruit-circuitpython-mcp230xx
  Found existing installation: sysv-ipc 1.0.0
    Uninstalling sysv-ipc-1.0.0:
      Successfully uninstalled sysv-ipc-1.0.0
  Found existing installation: spidev 3.4
    Uninstalling spidev-3.4:
      Successfully uninstalled spidev-3.4
  Found existing installation: Adafruit-PureIO 0.2.3
    Uninstalling Adafruit-PureIO-0.2.3:
      Successfully uninstalled Adafruit-PureIO-0.2.3
  Found existing installation: RPi.GPIO 0.7.0
    Uninstalling RPi.GPIO-0.7.0:
      Successfully uninstalled RPi.GPIO-0.7.0
  Found existing installation: Adafruit-PlatformDetect 1.3.4
    Uninstalling Adafruit-PlatformDetect-1.3.4:
      Successfully uninstalled Adafruit-PlatformDetect-1.3.4
  Found existing installation: rpi-ws281x 4.2.2
    Uninstalling rpi-ws281x-4.2.2:
      Successfully uninstalled rpi-ws281x-4.2.2
  Found existing installation: Adafruit-Blinka 3.0.0
    Uninstalling Adafruit-Blinka-3.0.0:
      Successfully uninstalled Adafruit-Blinka-3.0.0
  Found existing installation: adafruit-circuitpython-busdevice 4.0.0
    Uninstalling adafruit-circuitpython-busdevice-4.0.0:
      Successfully uninstalled adafruit-circuitpython-busdevice-4.0.0
  Found existing installation: adafruit-circuitpython-mcp230xx 2.2.0
    Uninstalling adafruit-circuitpython-mcp230xx-2.2.0:
      Successfully uninstalled adafruit-circuitpython-mcp230xx-2.2.0
Successfully installed Adafruit-Blinka-3.0.0 Adafruit-PlatformDetect-1.3.4 Adafruit-PureIO-0.2.3 RPi.GPIO-0.7.0 adafruit-circuitpython-busdevice-4.0.0 adafruit-circuitpython-mcp230xx-2.2.0 rpi-ws281x-4.2.2 spidev-3.4 sysv-ipc-1.0.0
pi@NASA1:~/NASA $ sudo python3 starlink.py
Traceback (most recent call last):
  File "starlink.py", line 45, in <module>
    mcp = adafruit_mcp230xx.MCP23017(i2c)
AttributeError: module 'adafruit_mcp230xx' has no attribute 'MCP23017'
pi@NASA1:~/NASA $ 
`

tinystarship, 2019-11-01:
as usual right after I post I find a solution.

I was looking at the release notes and it mentioned 

> NOTE import syntax is changed - see examples

then I check an example

> import board
>import busio
>import digitalio
>
>from adafruit_mcp230xx.mcp23008 import MCP23008

I switched to that and was able to get past the error. 

This is the guide I was following originally: 
https://learn.adafruit.com/using-mcp23008-mcp23017-with-circuitpython/python-circuitpython

Thanks!

closed 2019-11-01 by tinystarship

caternuson, 2019-11-01:
Thanks for point that out. The guide has been updated.

ladyada, 2019-11-01:
thank you @caternuson !