adafruit/Adafruit_CircuitPython_MCP230xx #23: Making the initialization of MCP23017 class optional?

postal, 2020-03-03:
https://github.com/adafruit/Adafruit_CircuitPython_MCP230xx/blob/31b582506b3a0813be43d5ed8ec027ba7ea1fb99/adafruit_mcp230xx/mcp23017.py#L65

I have connected an MCP23017 to my Raspberry Pi. During reboot the MCP23017 is used to configure the inputs, outputs and pullup resistors. If I use the library afterwards, the configuration is reset when the class is initialized. 

Is there a way to bypass the initialization?

dherrada, 2020-06-25:
Maybe try something like what's discussed here:
https://stackoverflow.com/questions/38010391/share-python-objects-between-two-or-more-py-files

tannewt, 2020-06-26:
I'd suggest we add a `reset=True` kwarg that can be used to disable the reset on startup. This will be needed when CircuitPython can deep sleep and therefore the code.py runs from scratch each time.

dherrada, 2020-06-30:
@tannewt Ok, I'll take a look at doing that

dherrada, 2020-06-30:
hi @postal Can you share the code you're using? I'd like to be able to test this with that exact code so I can be sure what you're trying to do will work.

six-zdpol, 2020-07-03:
hi @dherrada, this is the script which is executed by cronjob after a restarting the Pi, to setup the mcp23017.

`#!/bin/bash
echo "setup all gpio on bank A to output (0x00)"
sudo i2cset -y 1 0x20 0x00 0x00
sudo i2cget -y 1 0x20 0x00

echo "setup all gpio on bank B to input (0xff)"
sudo i2cset -y 1 0x20 0x01 0xff
sudo i2cget -y 1 0x20 0x01

echo "setup pullup resistors on bank B (0xff)"
sudo i2cset -y 1 0x20 0x0D 0xff
sudo i2cget -y 1 0x20 0x0D
`

All these settings are reset when the library is initialized.

postal, 2020-07-03:
hi @dherrada, this is the script which is executed by cronjob after a restarting the Pi, to setup the mcp23017.

`#!/bin/bash
echo "setup all gpio on bank A to output (0x00)"
sudo i2cset -y 1 0x20 0x00 0x00
sudo i2cget -y 1 0x20 0x00

echo "setup all gpio on bank B to input (0xff)"
sudo i2cset -y 1 0x20 0x01 0xff
sudo i2cget -y 1 0x20 0x01

echo "setup pullup resistors on bank B (0xff)"
sudo i2cset -y 1 0x20 0x0D 0xff
sudo i2cget -y 1 0x20 0x0D
`

All these settings are reset when the library is initialized.

dherrada, 2020-07-03:
@postal Great. Thanks. I'll load this on my pi and try to add the reset keyword

JuliaHorvath, 2021-01-13:
@dherrada Hey, I am having the same problem. Did you find any solution yet? :)

dherrada, 2021-01-14:
@JuliaHorvath Hey. I don't think I ever ended up getting to it. Is there a way to set those settings after the library has been initialized? If there is, that would probably be a pretty simple workaround.

@tannewt Is this still something you think I should do? If so, I can add to my list

tannewt, 2021-01-14:
@dherrada Yup! We should still do it. We actually have sleep support now so this will be even more useful.

ladyada, 2021-01-15:
@JuliaHorvath @postal can ya try the PR ?

jposada202020, 2021-05-12:
@JuliaHorvath @postal Let me know if you tried the PR. I think we are good to close this issue. Thanks.

JuliaHorvath, 2021-05-13:
yes, it is working now! Thanks!

jposada202020, 2021-05-13:
@JuliaHorvath Thanks :)

closed 2021-05-13 by jposada202020