adafruit/Adafruit_CircuitPython_MCP230xx #27: Interrupt capability

ib101, 2020-06-19:
Dear Adafruit,
Is Interrupt capability already supported? I see all all kinds of fuctions in the mcp23017.py code but the website says no interrupt support is here yet. Would also be great to add a little example code. Thanks!

dherrada, 2020-06-25:
CircuitPython doesn't support the use of interrupts, although this library definitely appears to use them, so I'm not really sure what's going on there.

siddacious, 2020-06-25:
This library like some others allows for setting up interrupts within the device and polling for their status by looking at the interrupt status register(s). This is essentially the same as polling an interrupt pin but instead asking the chip's register(s) directly.

@ib101 There are two examples using the interrupt capability such as it is:
https://github.com/adafruit/Adafruit_CircuitPython_MCP230xx/blob/master/examples/mcp230xx_event_detect_interrupt.py
https://github.com/adafruit/Adafruit_CircuitPython_MCP230xx/blob/master/examples/mcp230xx_leds_and_buttons_irq.py

If those are not sufficient, please re-open this ticket and include a more specific example of what features or scenarios you would like to see in an interrupt example

closed 2020-06-25 by siddacious