adafruit/Adafruit_CircuitPython_MCP230xx #32: Change asserts to exceptions.

caternuson, 2020-07-21:
There are couple of places where asserts are being used to check values. Suggest changing those to an appropriate exception along with a informative message.

ex:
```python
assert 0 <= pin <= 7
```
change to:
```python
if not 0 <= pin <= 7:
    raise ValueError("Pin number must be 0-7.")
```

jerryneedell, 2020-07-21:
Is this a general desire across libraries? While the exceptions are far more user friendly, they do result in a significant increase in memory usage compared to an assert.

caternuson, 2020-07-21:
Good question. This sort of change has been done before, in other repos/libs. The main motivation is to provide a more useful message for the user. I guess another option is to add the message to the assert.
```python
assert 0 <= a <= 7, "Bad value"
```

tannewt, 2020-07-22:
> Is this a general desire across libraries? While the exceptions are far more user friendly, they do result in a significant increase in memory usage compared to an assert.

Why does it take more memory? The error string?

jerryneedell, 2020-07-22:
> > Is this a general desire across libraries? While the exceptions are far more user friendly, they do result in a significant increase in memory usage compared to an assert.
> 
> Why does it take more memory? The error string?

I think that is a lot of it -- but in my RFM9x works I found it to save a few 10s of bytes.  I can do a more thorough study of it.

jerryneedell, 2020-07-23:
@tannewt  Here is a quick example of the difference in memory usage:
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.1-233-ga96e00834 on 2020-07-23; Adafruit Feather M0 RFM9x with samd21g18
>>>
>>> import gc
>>> gc.mem_free()
21488
>>> pin = 1
>>> assert 0 <= pin <= 7
>>> gc.mem_free()
21168
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.1-233-ga96e00834 on 2020-07-23; Adafruit Feather M0 RFM9x with samd21g18
>>> import gc
>>> gc.mem_free()
21488
>>> pin = 1
>>> if not 0 <= pin <= 7:
...     raise ValueError("Pin number must be 0-7.")
...     
...     
... 
>>> gc.mem_free()
20976
>>> 
```

jerryneedell, 2020-07-23:
hmmm --
I ran this:
```
import gc
gc.collect()
print(gc.mem_free())
pin = 1
gc.collect()
assert 0 <= pin <= 7
print(gc.mem_free())
if not 0 <= pin <= 7:
    raise ValueError("Pin number must be 0-7.")
print(gc.mem_free())
gc.collect()
``` 
no difference ... I'm not sure why...
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.1-233-ga96e00834 on 2020-07-23; Adafruit Feather M0 RFM9x with samd21g18
>>> import test_assert
20848
20848
20848
>>> 
```

caternuson, 2020-07-23:
@tannewt Whichever way this goes, `asserts` or `Exception`s, it might be good to document the preference, or general guidance, in the CP Design Guide.

tannewt, 2020-07-23:
@jerryneedell by having the assert vs raise at the top level it gets thrown away after it's run. You'll want to wrap them in a function to compare.

How does the exception thrown change? I bet raising a specific exception is easier to understand for a user. (I think we're overusing RuntimeError generally.)

@caternuson Good idea!

caternuson, 2021-02-11:
@tannewt @jerryneedell What's current thinking on this? asserts or Exceptions?

jerryneedell, 2021-02-11:
@caternuson I have not done anything more with regards to this.  I suppose it may make sense to suggest that Exceptions are easier to understand and should be the Recommended approach but if memory is a critical issue, then using asserts may be an useful alternative.

tannewt, 2021-02-11:
I agree with Jerry, exceptions with good messages.

closed 2021-02-11 by ladyada

caternuson, 2021-02-11:
Fixed with #39 

For the discussion above about capturing all this in the Design Guide:
https://github.com/adafruit/circuitpython/pull/4185

commit c84c9012 (2021-02-11): Merge pull request #39 from SAK917/master

Replace assert with if statement (Fixes #32)