adafruit/Adafruit_CircuitPython_MCP230xx #50: MCP23S17 can't using spidev1.* OSError: /dev/spidev6.0 does not exist

RunoKim, 2022-01-12:
I already enabled raspberry pi 4 has 6 spi as below
pi@pi:~/Desktop $ ls /dev/spi*
/dev/spidev0.0 /dev/spidev0.1 /dev/spidev1.0 /dev/spidev1.1 /dev/spidev1.2 /dev/spidev5.0

by the way, spidev0.* and spidev5.* is can run, but if i try to execute spi w/ spidev1.* than "OSError: /dev/spidev6.0 does not exist" occurred. please see the below.

            import board
            import busio
            import digitalio

            from adafruit_mcp230xx.mcp23s17 import MCP23S17
            spi = busio.SPI(board.SCK_1, MOSI=board.MOSI_1, MISO=board.MISO_1)
            Traceback (most recent call last):
            File "", line 1, in
            File "/home/pi/.local/lib/python3.9/site-packages/busio.py", line 286, in init
            self._spi = _SPI(portId)
            File "/home/pi/.local/lib/python3.9/site-packages/adafruit_blinka/microcontroller/generic_linux/spi.py", line 22, in init
            self._spi = spi.SPI(device=(portid, 0))
            File "/home/pi/.local/lib/python3.9/site-packages/Adafruit_PureIO/spi.py", line 167, in init
            raise IOError("{} does not exist".format(device))
            OSError: /dev/spidev6.0 does not exist

caternuson, 2022-01-12:
You have 6 total ports, but none are the expected port ID of 6 when using those pins:
https://github.com/adafruit/Adafruit_Blinka/blob/f6a653e6cc34e71c9ef7912b858de1018f08ecf8/src/adafruit_blinka/microcontroller/bcm2711/pin.py#L82
BCM2711 (Pi4) SPI6 pin reference:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#spi6-bcm2711-only

How did you enable the other SPI ports?

RunoKim, 2022-01-12:
Thanks, I solved it.
i wanted to use 3 chip select (chip enable) pin on my Rpi4.
So i changed ".local/lib/python3.9/site-packages/adafruit_blinka/microcontroller/bcm2711/pin.py" as below.
**Original:**
# ordered as spiId, sckId, mosiId, misoId
spiPorts = (
    (0, SCLK, MOSI, MISO),
    (6, SCLK_1, MOSI_1, MISO_1),
    (2, SCLK_2, MOSI_2, MISO_2),
    (3, D3, D2, D1),
    (4, D7, D6, D5),
    (5, D15, D14, D13),
)
**Modified:**
# ordered as spiId, sckId, mosiId, misoId
spiPorts = (
    (0, SCLK, MOSI, MISO),
    **(1,** SCLK_1, MOSI_1, MISO_1),**
    (2, SCLK_2, MOSI_2, MISO_2),
    (3, D3, D2, D1),
    (4, D7, D6, D5),
    (5, D15, D14, D13),
)
And now i can using spi1 w/ 3cs pin.
Thanks again.

closed 2022-01-12 by RunoKim