adafruit/Adafruit_CircuitPython_MCP230xx #54: MCP23018/MCP23S18 Support

DonutCables, 2022-03-26:
I don't see this explicitly listed as supported, so I figured I'd ask. Given the current state of silicon in the world, these 018/S18 seem to be way more available, but info on using them seems to be what's rare.

Datasheet here [https://ww1.microchip.com/downloads/en/DeviceDoc/22103a.pdf](https://ww1.microchip.com/downloads/en/DeviceDoc/22103a.pdf)

From my uneducated glace between this and the 017/S17 docs, it looks largely the same for serial data aside from IOCON in the Control Registers: X17 has no bit 0 INTCC on either IOCON, X18 has no bit 4/bit 3 DISSLW/HAEN on IOCON. GPIO Port Registers show no differences. I'm not really sure if that's even relevant, but that's what I can pick out as a difference.

Unfortunately I have none on hand to just see if they work on X17 code, though I imagine there's details inside that need to be correct to work.

DonutCables, 2022-04-06:
Received my 018 chips and got them in a board to test. Looks like for the most part they "just work" with the 017 libs, the only thing that I found different is pins set to output need an explicit `Pull.UP` when initialized. I suspect this is the outputs defaulting to open-drain while the 017 doesn't. I'm sure more thorough testing would be good before declaring this fully supported, but it may at least be worth a mention somewhere in the docs that this can be done.

redonwhite, 2022-04-30:
Thanks for posting this, google brought me here with the same issue. Now I can confirm MCP23018 seems to work like the MCP23017 when `Pull.UP` is used on outputs.